victordiaz/PHONK #1: different behavior when it should be the same

danidask, 2017-12-30:
Hi,
I'm working in an app to upload a picture to a server, and I'm experimenting a weird behavior. In the flowing code, when I use filename inside data it crashes, but when I use filename2 it works flawlessly, but both should be the same. Is it something I did wrong or is a bug? (btw, when it crashes, how can i read the log or see whats going on?)  

Thanks Victor for your work, awesome project!

```
ui.addTitle(app.name)

var url = 'http://192.168.1.159:9000/addimage'

ui.addButton('Upload picture', 0.8, 0.6, 0.19, 0.09).onClick(uploadPicture)

function uploadPicture() {
  device.vibrate(500)
    
  var number = 1
  var prefix = 'pic'
  var extension = '.png'
  var filename = prefix + number + extension
  
  var filename2 = 'pic1.png'
  console.log(typeof(filename))
  console.log(typeof(filename2))
  console.log('**' + filename + '**')
  console.log('**' + filename2 + '**')
  if (filename === filename2){
    console.log("same")
  }
  else {
    console.log("different")
  }
  
  var data = [{ 'name' : 'userfile', 'content' : filename, 'type' : 'file', 'mediaType': 'image/png'}]

  network.httpPost(url, data, function (e) {
    // console.log(e)
    if (e.status == 200){
      ui.toast("picture uploaded")
      console.log("picture uploaded")
    }
    else {
      ui.toast("error uploading picture")
      console.log("error uploading picture")
    }
   
  })
}
```

victordiaz, 2018-01-02:
Hi @danidask, can you provide me in a gist a server code sample? 
I changed sometime ago the HTTP POST library and might be not well tested.

If you want to see the log, the best is to have ADB installed in your system
`adb logcat`

I will create a new issue to introduce a logging mechanism that writes up in a .log file

danidask, 2018-01-04:
Ok, this is the server's code, but you can test it even if there is no server listening.
https://gist.github.com/danidask/e6ad97fee659062cb5449d96038d1b85

victordiaz, 2018-01-15:
Apparently the Javascript Engine (Mozilla Rhino) has a different data type for a concat String. Adding toString seems to solve the problem 334bcc1d228214a5a382fb0266e28fcf598f3fe7

On the other side I included a logging mechanism https://github.com/victordiaz/phonk/issues/2

I'm going to fix a couple of things more and upload a new version in a couple of days. I will write you back here once ready :)

victordiaz, 2020-06-20:
I close the issue. There is a lot of rework in the last release, maybe hopefully was fixed? :)

closed 2020-06-20 by victordiaz