victordiaz/PHONK #4: Update the JS interpreter with something more modern

victordiaz, 2019-05-19:
Phonk uses Mozilla Rhino interpreter to execute the scripts as it binds very easily with Java.
Although it works quite nicely, it feels a bit old. There is no support of ES6 among other things.

Maybe somebody with experience out there? :)

victordiaz, 2020-06-20:
I did some experiments using [J2V8](https://github.com/eclipsesource/J2V8) and realized it would need a lot of effort to change it from Mozilla Rhino. 

I think for now is not worth the time, so I close the issue.

marclava, 2020-06-29:
It doesn't look like GrallJS (community version) is available for Android:
https://github.com/graalvm/graalvm-ce-builds/releases/tag/vm-20.1.0
But maybe it doesn't have to, I'm not a Java programmer...

Also, maybe Nashorn would be a good option (even if deprecated in JDK11):
https://en.wikipedia.org/wiki/Nashorn_(JavaScript_engine)
https://docs.oracle.com/javase/9/nashorn/JSNUG.pdf
https://wiki.openjdk.java.net/display/Nashorn/Rhino+Migration+Guide

danielzfranklin, 2020-07-01:
What about using Babel to transpile modern JavaScript to ES5? I think that's how greasemonkey solves this problem. Babel-standalone (https://babeljs.io/docs/en/babel-standalone) can run on a JavaScript interpreter and transform modern JavaScript into older JavaScript

victordiaz, 2020-07-01:
@danielzfranklin that sounds very interesting. 
I thought babel could only run using node but the standalone version could do the work!

I'd be happy to include it if it works

victordiaz, 2020-07-02:
I did a small test yesterday I couldn't use babel-standalone. When I load the library Rhino is kinda happy about it but then I dont see the Babel object instantiated.
Researching a bit in the internet, I did not find any person using it with Rhino :/

Piroro-hs, 2020-07-04:
How about using Babel on the web-editor side? It works perfectly on modern browsers.

victordiaz, 2020-07-04:
@Piroro-hs That's a quite interesting idea.
I'm wondering if it will become more difficult to share scripts among people, since you always have to transpile it with the Web Editor...

Maybe a good compromise is to be able to write a main.js.es6 in the editor. Then the editor using Babel will send the transpiled "main.js" to the device, so everything will be normal from the perspective of the app.