victordiaz/PHONK #9: Create guide to compile projects into APK

victordiaz, 2019-05-19:
It is possible to wrap the scripts into an APK file for easy distribution. Not easy though.
If there is interest I will write a guide.

micfio, 2019-12-12:
Can provide me a detailed guide in order to convert phonk script to apk ?Thanks In Advance

w-A-L-L-e, 2020-07-02:
Indeed this is only feature missing to make this a really nice useful app. Once you've got something that works for you it would be awesome to make it into a standalone apk to share...

victordiaz, 2020-07-02:
I've seen an app called APDE that compiles on device. Which is super cool [https://github.com/Calsign/APDE](https://github.com/Calsign/APDE)
I think is a bit out my action range for now, but it would be very nice to have something like that :)

I'll write the "compiling guide" using a computer after I release the next version.

BanDroid, 2024-03-27:
> I'll write the "compiling guide" using a computer after I release the next version.

when?