victordiaz/PHONK #12: Create tutorials and/or showcases

victordiaz, 2019-05-21:
It would be great to create a few tutorials or script showcases that people can learn from and get inspired.

I could suggest something like:

- Simple videogame
- Musical instrument
- Small robot with an Arduino attached
- Controlling lights using MQTT

burnedsap, 2020-07-13:
I can help with some tutorials on using sensor data to create Processing sketches. I do need a little help on figuring out a few things though.

victordiaz, 2020-07-14:
That would be great! Let me know how can I help you :)

b2renger, 2020-07-14:
I can also help if needed !

burnedsap, 2020-07-14:
Amazing, is there a discord/slack where we can coordinate?

victordiaz, 2020-07-15:
Hi, I just created a discord here [https://discord.gg/Rt2mkWp](https://discord.gg/Rt2mkWp)
Feel free to join :)