victordiaz/PHONK #13: Spreading the word about Phonk

victordiaz, 2019-05-21:
It would be nice so spread a bit the word about Phonk using certain websites and social networks. I'm not the greatest doing PR 😅 and somebody willing to help me on that would be supernice :)

ffranchina, 2019-10-09:
Hello @victordiaz! Thanks for working on this nice project!
I was a user of Protocoder some time ago and now I saw that the project has kind-of been shut down. It took me a lot to find Phonk, though: it has the same author (you), same aims, same UI but they appear as completely different projects and there's no link between the two.
As a first step for advertising Phonk, I would propose you to redirect all the users are looking for Protocoder to this new platform.
I cannot guess if you have your reasons for not having done it or it was simply lack of time. In the latter case I would be happy to help you in the process of migration!
Have a good day!

victordiaz, 2019-10-16:
Hi @ffranchina, thanks for using the project :)
Thanks for your suggestion! I'm finishing about to release the version 1.1 and right after it I will dedicate some time with the websites :)
- For now my first priority is to upload the project to F-droid and Google Play store. This later one has less priority since the last changes in the Play Store is forcing me to change a bit Phonk.
- And second priority is to setup a CI workflow.
- After that I would love to spread the word about it :) 

If you can help me with any of those points now I would be more than glad!

victordiaz, 2020-07-12:
@ffranchina It took me like forever but now there is a message in protocoder repo and website inviting to go to PHONK :)