victordiaz/PHONK #17: Add MQTT documentation

victordiaz, 2020-06-17:


garydion, 2020-07-03:
Yes please!  I'm trying to get the example to work this evening and not having much luck.  This is the first example in Phonk I have run across that didn't just work (So kudos for that!).  But then, this is also the one example I've modified most heavily, and I'm new to JavaScript, so it's likely I broke something.  A little more documentation might help me out of this NullPointerException hole I've dug.  Thanks!

garydion, 2020-07-03:
I've been playing a bit more and the MQTT functions work great with my local Mosquitto broker.  I can subscribe and publish to feeds.  But for some reason it doesn't seem to like the Adafruit.io broker, at least not with any of the permutations I've tried so far.  None of the MQTT commands (connect, subscribe, or publish) generate an error, so I'm afraid I don't have any debug info to offer.

victordiaz, 2020-07-04:
Hi @garydion 

There was a pull request fixing an issue with MQTT. [https://github.com/victordiaz/PHONK/pull/35](https://github.com/victordiaz/PHONK/pull/35)

The version 1.2.5 is up (Google Play takes a bit longer to update). 
Could you try to see if the error is fixed now?

The example has changed a bit, so after updating the app you should go to Settings -> Reinstall Examples. Then Go to Network -> MQTT to see the new example (basically changes the field username, now is named just ´user´)

garydion, 2020-07-04:
It works great now, even on Adafruit.io!  Thanks!  AND the buttons show text as they should.  

The pull was from a good friend of mine who knew a LOT more about it than I did.  And while I thought my Mosquitto broker was checking name/password, turns out it wasn't all along.  So that's why it was working despite the bug.

victordiaz, 2020-07-04:
Cool to hear!
I changed not so long ago the underlying MQTT library because the old one had some problems, But it seems I didnt test it enough :)
I will keep the issue open until I add the methods to the documentation

42loop, 2020-07-08:
hello, 

for me the mqtt example still does not work:

Wrapped java.lang.NullPointerException: Attempt to invoke virtual method 'char[] java.lang.String.toCharArray()' on a null object reference (MQTT#22)

tried different mqtt brokers, same on all, even compiled from git in android-studio...

victordiaz, 2020-07-08:
Hi @42loop 
Did you update the examples in Settings -> Reinstall Examples?

42loop, 2020-07-08:
yes, removing user and password in connectiondata helps
thanks

victordiaz, 2020-07-08:
@42loop Thanks for reporting the issue. I'll update the MQTT example in the next release :)

42loop, 2020-07-14:
hello,
in the 1.2.7 release the mqtt connect and subscribe functions work, but publish does not return, have to kill phonk on the device
greetz

victordiaz, 2020-07-14:
Hello @42loop 
What do you mean with "publish" does not return?
Could you elaborate a bit more?

Thanks!

42loop, 2020-07-14:
i mean, if i place a console.log() after the publish command, it never gets called.
also the button stays in 'pressed' state (gray)

victordiaz, 2020-07-15:
@42loop I see! Maybe the paho library I'm using has some blocking mechanism. I'll have a look thx :)

victordiaz, 2020-07-15:
@42loop 
I just had a look and couldn't replicate the problems.
I'm using mosquitto broker for testing with the PHONK's MQTT built-in example. I just added a console log right after calling the publish method.
The console log appear and the button behaves normally. 

Could you send me the details of your code & setup so I try to replicate it?

Thanks!

garydion, 2020-07-16:
For what it's worth, moving from 1.2.5 to 1.2.7 didn't break my ability to use either Adafruit.io or CloudMQTT.

victordiaz, 2020-07-16:
@42loop I can see now what you mean. It seems that when it cannot connect or deliver the message the UI getz frozen. I'll try to fix it ASAP 

@garydion thx for checking it out

victordiaz, 2020-08-05:
@42loop I just uploaded a new version 1.2.8 and I think it solves the issue with the blocking stuff.
I hope it didn't break anything else :)

42loop, 2020-08-06:
in 1.2.8 connect and publish works (as i can see in a different client) but in phonk onNewData() never gets called.
the other client tells me that subscribe() does not work.

42loop, 2020-08-06:
correcting myself: onNewData() gets called,
but e is always {"status":"deliveryComplete"} instead of the message

victordiaz, 2020-08-06:
@42loop I just found the problem.
Since I changed to asynchronous methods, when `c.subscribe('debug')` is called, the connection is not ready yet.

I will change the example tonight to fix it. Thanks for letting me know!

victordiaz, 2020-08-06:
Hi all,

Since I moved MQTT to work asynchronously I had to change a tiny bit the API. The main problem was that the connection had to be established before subscribing to a topic.

I uploaded here the new example 
[https://github.com/victordiaz/PHONK-examples/blob/master/examples/Network/MQTT/main.js](https://github.com/victordiaz/PHONK-examples/blob/master/examples/Network/MQTT/main.js)

which will be included in the version 1.2.9 in a couple of days. If you have some comments on the new example/API feel free to do it!

Hopefully things wont have to change anymore after this :)