victordiaz/PHONK #19: Add examples to assets folder

marclava, 2020-06-26:
I was able to compile the APK, but the examples are not included in the source code (in the assets folder, sot the compiled app has no examples available. Either include the examples to the source code or provide a method to add examples from a remote computer.

victordiaz, 2020-06-26:
Hi @marclava 
Thanks for pointing that out. It seems that I was pushing the examples to the wrong origin 😅
The examples are here [https://github.com/victordiaz/phonk-examples](https://github.com/victordiaz/phonk-examples)

I updated the [README](https://github.com/victordiaz/phonk) with instructions on how to build the project, since you have to add as well the web editor to the android project.

marclava, 2020-06-26:
Thanks for the precision and the README update; keeping parts in different git repos is fine, and maybe you can include them to the main app repo as git submodules. 

I have so many questions and comments about Phonk that it'd be nice to have a place to chat about it (because this issue reporting system may not be the ideal communication channel). For example, I want to learn how to include new features to Phonk, like video streaming; I'm not (yet) familiar with Android development with Java, and how to include compiled libraries and interface them to the Rhino Javascript engine.

victordiaz, 2020-06-26:
I'm happy to have a chat using a different channel, feel free to send me an e-mail and we figure it out. Just git log in the repo to see my mail.

victordiaz, 2020-07-02:
I close this issue since it was already resolved

closed 2020-07-02 by victordiaz