victordiaz/PHONK #20: PHONK for background automation

victordiaz, 2020-06-29:
It was suggested if PHONK could automate things based on system events. I will play with the idea and open this issue to keep track of things

gtusr, 2022-08-06:
Any progress yet on this front?

victordiaz, 2022-08-08:
Hi @gtusr

Not really. I fixed some stuff to better run background scripts like a year ago but thats all.
In theory lots of things could be done already. Is there something specific that you would like to do?

gtusr, 2022-08-11:
I was just wondering if PHONK could someday become a replacement for Tasker since PHONK can already respond to various "profile triggers," as demonstrated on the project homepage.

I'm not familiar enough yet with the Android framework to estimate how much work would be required to re-create a "Tasker-like" engine that could manage several profiles simultaneously for the user in the background, but this is why I asked about whether or not developing a framework for managing background processes, that can balance issues like battery usage with responsiveness of the phone user interface during foreground activities, is on the future roadmap for PHONK.

victordiaz, 2022-08-11:
I think lots of things can be done already.

For example mixing this 3 examples:

1) create a background running script with this https://github.com/victordiaz/PHONK/tree/develop/PHONK-examples/Advanced/Background%20App
2) Get the battery status with this https://github.com/victordiaz/PHONK/tree/develop/PHONK-examples/Device/Info
3) And depending on the battery status you could send a MQTT message to a broker https://github.com/victordiaz/PHONK/tree/develop/PHONK-examples/Network/MQTT or send a HTTP GET request https://github.com/victordiaz/PHONK/tree/develop/PHONK-examples/Network/HTTP%20GET

I will be happy to help if somebody has some needs that PHONK does not provide yet. We can try to figure out the best and easiest way to accomplish it :)
So if you or anybody else have some need or suggestions that we could add to the baseline we already have I'm all ears :)

If you have any question feel free to ask. I know things are not very well documented, so the best way is running scripts and see how they work. I tried to keep all the examples small so they can be understood.