victordiaz/PHONK #21: title in buttons is not displayed

boriskohen, 2020-06-29:
it's appears only on devices with kitkat (android 4.4).
affected 1.2.0 and 1.2.2 versions. affected installation wizard and projects view.
[image]

victordiaz, 2020-06-29:
Oh! thats interesting, I will have a look today.
What is the screen size? I will try to replicate it with an emulator

boriskohen, 2020-06-29:
320x480px  
3" size

victordiaz, 2020-06-30:
Hi @boriskohen

I think I fixed it, it seems that Android 4.4 didn't pick the "backgroundTint" property, so I changed to the standard "background"
I will do a new release this week with the changes

I will let you know when the new version is online so you can double check and close the issue

Thanks for pointing it out!

tinkercnc, 2020-07-02:
Same behavior on:
device
    type: phone
    manufacturer: OnePlus
    model name: A0001

screen
    brightness: 93
    screen: true
    orientation: portrait
    screen dpi: 480
    screen resolution: 1080 x 1920

Cyanogen OS
Android 4.4.4 

Just FYI.

garydion, 2020-07-02:
I can also confirm it occurs on an Alcatel OneTouch running 4.4.4.  Additionally, the proceed buttons during installation were invisible.  These were the ones which were to say "Let's Go", "Ask Permissions", and "Install Examples".  But thanks to your "Getting Started" guide on the website, I knew where to tap.  Despite the bug, I'm already having a blast!

victordiaz, 2020-07-03:
I just uploaded a new version, cand you check if buttons appear now? thanks :)

tinkercnc, 2020-07-03:
> I just uploaded a new version, cand you check if buttons appear now? thanks :)

Nope, for instance the    
Phonk Example Device info shows this:

[image]

garydion, 2020-07-03:
I can confirm that the buttons now display correctly during install.  Though like tinkercnc, my application buttons are still all white like in his screen shot.

victordiaz, 2020-07-03:
Ok, it seems that there is something wrong :)
I have a device with 4.2 and dont get that problem :/ 
I'll have a deeper look tonight with the emulator

Thanks for checking

victordiaz, 2020-07-03:
I think I found the problem. Apparently devices < 5.0 don't like background with transparencies. So I made a small hack to use a color shader on Androids lower than 5.0

I uploaded a version here
[https://github.com/victordiaz/PHONK/releases/tag/1.2.4](https://github.com/victordiaz/PHONK/releases/tag/1.2.4)

Could you check it? Thanks :)

tinkercnc, 2020-07-03:
Yeah! 
Yes it's working!
Great piece of software!

Thanks a lot!

boriskohen, 2020-07-03:
Thank you. It's working now. Great piece of software!

closed 2020-07-03 by boriskohen

garydion, 2020-07-04:
Yep, fixed on my phone too.  The buttons looks great.  Thanks!