victordiaz/PHONK #23: Edit Script Crashes the entire App for me

st0le, 2020-06-30:
I'm guessing it should pop up an app chooser to edit the file? But it seems to crash the app.

Using the Web Editor is a workaround.

st0le, 2020-06-30:
I tried it a couple of hours later and it seems to work fine. I don't know what fixed it.

Considering I can't repro, perhaps this can be closed.

victordiaz, 2020-06-30:
Hi @st0le,
1. Could you tell me about your device? Model, android version, etc?
2. Did you try to access the Editor right after installing the app?

st0le, 2020-06-30:
1. Pixel 2 (small screen), Android 10.
2. Yes. I checked a couple of examples and then tried to view the code.

I tried clearing the cache/storage and could not repro.
I tried uninstall/install and still could not repro it.

victordiaz, 2020-06-30:
ok! I see!
I asked you about the model and if it was the first install because it happened to me the same once but then I was never able to replicate it.
I will close the issue then and if I ever see the bug again I will reopen it :)

Thanks for the info @st0le

victordiaz, 2020-07-03:
I've had the same problem using an emulator.

It seems that the WebView component sometimes is not present in some devices > 5.0
[https://stackoverflow.com/questions/46048912/android-failed-to-load-webview-provider-no-webview-installed](https://stackoverflow.com/questions/46048912/android-failed-to-load-webview-provider-no-webview-installed)

I will remove the webview reference, since it's a bit of legacy code and is not needed anymore. It should work fine in the next version

victordiaz, 2020-07-03:
I think the About section should move to a real view rather than a WebView to prevent this problem too

BillisC, 2020-07-04:
Hi guys, I have the same issue on Redmi Note 8 (Android 10)
The app crashes both when I run a script (on web) and when I open it via the app.

I'm not sure if it's helpful but here is the crash log:
https://pastebin.com/acSt8gJW

I should also note that I gave permissions before doing anything after installing. Also tried clearing cache after the update.

victordiaz, 2020-07-04:
Hi @BillisC 

Few questions so we can try to solve the mistery :)

- Does it happen with any of the examples?
- Do you use an alternative rom?
- Which version of the phone is it? European / Indian / Chinese / ... ?

BillisC, 2020-07-04:
> Hi @BillisC
> 
> Few questions so we can try to solve the mistery :)
> 
> * Does it happen with any of the examples?
> * Do you use an alternative rom?
> * Which version of the phone is it? European / Indian / Chinese / ... ?

1. Yes it happens with every script I'm trying to run(incl. examples and my new files)

2. Yep i use custom rom, but it's fully stable and I never had issues with any apps but still it could be a possible reason.

3. It shouldn't matter since I'm in a custom rom but it came with global version(For all countries except China and India I think)

victordiaz, 2020-07-04:
Hi @BillisC,

I think the error is a bit different than this isue, so I created a new one. 
We can keep on going with the conversation here #36

victordiaz, 2020-07-06:
I think this issue is resolved

closed 2020-07-06 by victordiaz