victordiaz/PHONK #32: APK 1.2.x do not install on Android 4.1.2

SuperUserNameMan, 2020-07-03:
Hello,

I've just discovered your project which seems quite promising.

I'm testing it on an Acer Iconia B1-A71 tablet with Android 4.1.2. (CPU: MTK8317T dual-cores 1.2GHz ; RAM : 512MB ; Display : 1024x600 ).
Unfortunaltely, none of APK 1.2.x could be installed on this device.

Only APK 1.0.0 installed and works with some bugs, errors and a few crashes when I test the examples.

If you want to extend the compatibility down to Android 4.1.2, or 4.1.1 or even 2.3.3, I could make some more tests for you if you need :-)

victordiaz, 2020-07-03:
hi @SuperUserNameMan,

I'd be very happy. I think it can go to 4.1 easily but I doubt it would go lower than that. PHONK depends on the appcompat libs which requires > 4.1.
But as I said, any help on this regard is more than welcome. I would love to go to the lowest version possible :)

Could you tell me what happens when you try to install it in 4.1.2?

victordiaz, 2020-07-03:
hi again @SuperUserNameMan ,

I managed to lower the version to 4.1.
Unfortunately there are a few things that dont work such LibPd and Processing :/

I think its worth to allow 4.1 devices use PHONK, but probably will need some warnings when trying to use certain features

SuperUserNameMan, 2020-07-03:
> Could you tell me what happens when you try to install it in 4.1.2?

Hello again @victordiaz ,

I've tested the latest APK 1.23 and pre-release APK 1.2.4 too : Same error when trying to install.

The error message is in french : "Erreur d'analyse : Un problème est survenu lors de l'analyse du package."
which means : "Analysis error : A problem occurred during the analysis of the package."

After a little web-search, the english version of this error might be "Parse error : there was an error parsing the package" (but i'm not sure).

When trying to install the APK, I can choose to install it using the "installation kit program" or the "Verify and install" (with a Google icon). I tried both, it made no difference.

SuperUserNameMan, 2020-07-03:
> I managed to lower the version to 4.1.
> Unfortunately there are a few things that dont work such LibPd and Processing :/
> 

With version 1.0.0, the Processing example crashed.

Also, there was no text displayed inside the example with the textbox when I was typing on the keyboard. (but maybe you fixed that with 1.2.4)

victordiaz, 2020-07-03:
Thanks for the feedback!

I haven't included the 4.1 changes in the 1.2.4 prerelease but I will include them in the final release which I will do tonight or tomorrow morning. I will let you know, so you can try!

victordiaz, 2020-07-03:
Just uploaded the version 1.2.4 to Github. You should be able to install it now. Let me know if it does work for you

SuperUserNameMan, 2020-07-04:
Great ! it works ! I'm all exited now ! :-D

As you can see with the screenshot below, the text into the yellow rectangle box at the top is too small. On the tablet, it is unreadable (  screen is 160dpi with a resolution of 600 x 976. ).

[image]

Here are the feedback of all my basic tests :

1) example ``Advanced / Capture Notification`` displays this error dialog : ``Wrapped android.content.Activity/NotFoundException : No Activity found to handle Intent { act=android.settings.ACTION_NOTIFICATION_LISTENER_SETTINGS } (Capture Notification)``

2) example ``Advanced / LiveCoding`` : when pressing Ctrl+Shift+X from the Web Editor, there is a very tiny unreadable line of text displayed temporarily at the bottom of the application on the tablet.

3) example ``GUI / Maps (OpenStreetMaps)`` : It does not display any map. The background is a gray grid with a "MAPS (OPENSTREEMAPS)" title in bold yellow,  a red circle, a cyan circle, two blue segments and a pinpoint-location icone. Zooming and scrolling moves them randomly.

4) example ``Media / Camera`` : it crashes PHONK. I tried with 'front' instead of 'back' since this tablet only has 1 camera, and it crashed too.

5) example ``Network / BluetoothLowEnergy`` : displays an error box at the bottom : ``TypeError : Impossible to call the method "{1}" of {0} (BluetoothLowEnergy:#17)`` (but my tablet does not have BLE)

6) BTW, the text into the kind of error box is very small an almost unreadable (but i've just noticed they appear into the Console box of the WebEditor too).

7) when testing ``Network / HTTP Server`` a error box appeared after i connected to the ``/helloworld`` page : ``error : Method 'io.phonk.runner.apprunner.api.PUI.toast(string.number)' can't be found (HTTPServer#24)``. And once I leave this example, a dialog popup "The application PHONK was stopped".

8) example ``LibPd`` crashed PHONK. (it worked on version version 1.0.0)

9) example ``Processing`` crashed PHONK.

What I could not test :
- vibration example 
- MQTT
- NFC
- mDNS
- IOIO, MIDI and Serial Comm 
- most sensor examples 

Regarding the Web Editor :
- when opening the "GUI/Maps" example under Firefox 78 on a 1600*900 display, the list of files into the FILES box on the right does not scroll :
[image]

This is a great application that you have made here !
Even with the little issues I listed above, it already gives a lot of potential to a lot of old devices.
I always to learn coding for android, but the java and Android SDK always made give up and run away. But now, with PHONK, I'm very excited to test it in more depth ! Thank you for publishing it !

victordiaz, 2020-07-04:
@SuperUserNameMan thanks for your exhaustive report!
This helps a lot to make it more backwards compatible.
I will go carefully through the list and create different issues for each problem.

victordiaz, 2020-07-08:
Hi @SuperUserNameMan,
I divided the issues in different ones as you can see linked, I will close this issue as it is resolved and continue with the other ones, feel free to follow the status of them. Thanks again for the exhaustive report!

closed 2020-07-08 by victordiaz

SuperUserNameMan, 2020-07-09:
@victordiaz : when you need me to do specific tests on my device, feel free to summon me :-)

victordiaz, 2020-07-09:
@SuperUserNameMan thanks. I will :)