victordiaz/PHONK #33: Lat and Lon mixed up on adding map marker

Slord6, 2020-07-03:
https://github.com/victordiaz/PHONK/blob/9ae6a710ff830bab45be1da1e65c80a27b2c2d2a/phonk_apprunner/src/main/java/io/phonk/runner/apprunner/api/widgets/PMap.java#L283

The Lat and Lon parameters are backwards in this initalisation.
Docs [ref](https://osmdroid.github.io/osmdroid/javadocAll/org/osmdroid/util/GeoPoint.html)

Simple Repro:

``` Javascript
/*	
 *  Description Live map
 *  by Sam :)
 */

ui.addTitle(app.name)

console.log("Hello!");

var map = ui.addMap(0, 0, 1, 1);
map.center(1, 12);
map.zoom(3);
map.showControls(true);

var marker = null;
sensors.location.onChange(function (data) {
  console.log("chg");
  if(!marker){
    console.log("new");
    marker = map.addMarker({
      lon: data.longitude,
      lat: data.latitude
    });
  } else {
    console.log("update", data);
    marker.position(data.latitude, data.longitude);
  }
  console.log("marker", marker);
});

sensors.location.start()
console.log("sensor started");
```

victordiaz, 2020-07-03:
You are totally right :)
I'll fix it and add it in the release of today, thanks for pointing it out

closed 2020-07-03 by victordiaz

victordiaz, 2020-07-04:
Version 1.2.5 is already up and with the fix :)

Slord6, 2020-07-06:
Great, thanks!

commit f3ea2cd8 (2020-07-03): lat, lon mixed up, this fixes #33