victordiaz/PHONK #34: Uncomfortably slow animation for new views

wottomgauss, 2020-07-03:
This duration + delay timing is uncomfortably slow for some purposes, like when user is expecting a fast UI update with new buttons and text. Would appreciate some way to adjust these values.

https://github.com/victordiaz/PHONK/blob/d7bb34f68d8a15baf17c392fc6c09b5f711531da/phonk_apprunner/src/main/java/io/phonk/runner/apprunner/api/PViewsArea.java#L162

victordiaz, 2020-07-03:
Hi @wottomgauss 

I see your point
I'm adding some way to easily customize the main theme. I can add a property to disable the animations if that works for you. It could be something like 

```
ui.theme({
  animations: false
})
```

If more people find it uncomfortable then maybe it could be just turned off by default

wottomgauss, 2020-07-03:
Thank you! Absolutely love the app so far, please keep up the good work

victordiaz, 2020-07-03:
@wottomgauss 

I removed the animation and it looks good!
So I'll remove the animation by default. It will be possible to reenable it with the animationOnViewAdd property when setting a theme.
This is an example:

``` js
ui.setTheme({
  background: '#000055',
  primary: '#FF0000',
  secondary: '#00FF00',
  textPrimary: '#FFFF00',
  animationOnViewAdd: true
})
})
```

victordiaz, 2020-07-03:
Just uploaded APKs in Github and pushed to Google Play (it will take a bit to be online). Have a look and let me know if that works for you :)

wottomgauss, 2020-07-05:
Yes this is working perfectly now, many thanks!

closed 2020-07-05 by wottomgauss