victordiaz/PHONK #36: Crash initializing NFC

victordiaz, 2020-07-04:
**Problem description**
Crash on script launching
[https://pastebin.com/acSt8gJW](https://pastebin.com/acSt8gJW)

**Device model / Android Version**
   Redmi Note 8 (Android 10 custom rom)

**Steps to reproduce the bug**
Launch a script from the device or the Web Editor

victordiaz, 2020-07-04:
@BillisC
There is a bit of old code in PHONK.
Each time an script launches the NFC tries to be initialized in devices supporting it. Automatic initialization is not good :/

Maybe the custom rom does not support NFC and thats why it crash.
Nevertheless, the NFC should not be initialized each time so this needs a fix asap

BillisC, 2020-07-04:
What's strange is that redmi note 8 does not have NFC unlike redmi note 8T. So I used an app that shows support and it indeed showed that it supports. I bet it's because the build is also made for 8T so it's probably a rom issue. Maybe it could also be what you said about old code

victordiaz, 2020-07-04:
@BillisC 
There is an alpha version here [https://github.com/victordiaz/PHONK/releases/tag/1.2.6-alpha](https://github.com/victordiaz/PHONK/releases/tag/1.2.6-alpha) that should correct the problem.
Do you mind having a look?

BillisC, 2020-07-04:
Wow everything works fine now. Small issue that it shows a random crash when i change between scripts. But the app doesn't crash and i can continue using it. 

https://pastebin.com/0WCcSuCS

victordiaz, 2020-07-04:
I just uploaded a new APK in the same link as before (1.2.6-alpha-2)
It should work now :)

I will keep this issue open until it the NFC stuff can be tested with some real NFC tags :)

victordiaz, 2020-07-04:
wait! I was too fast :)
Check the alpha3.... :)

BillisC, 2020-07-04:
Alpha 3 works perfectly 👍

BillisC, 2020-07-04:
One thing I noticed is that the nfc example led to an app crash but would it be possible to show a text like "Unsupported" or similar instead?

victordiaz, 2020-07-04:
Yes, that would be good.
I dont have the right hardware but if you can test it for me I'd be happy to do it

BillisC, 2020-07-04:
sure thing, right hardware you mean no nfc hardware right?

victordiaz, 2020-07-14:
Hi @BillisC 
There is a new version 1.2.7 online. Can you check if the same thing is still happening? Thanks

BillisC, 2020-07-14:
The nfc is fixed and the app works. Not sure if you changed the example nfc since it crashes the app

victordiaz, 2020-07-14:
Hi @BillisC 
I don't understand very well. Does it crash on your device?
In my PIxel 3a NFC works both reading and writing with 1.2.7

BillisC, 2020-07-14:
Let me explain,
My phone (redmi note 8) does not support NFC. The app used to crash because of some old code that you already fixed. Now everything is fine except from the NFC example script which instantly crashes the app because no NFC is found. So my idea was to add a message like "NFC is unsupported" in it  instead of crashing the app.

victordiaz, 2020-07-14:
Oh, I see! :)
In theory I'm checking if th NFC is supported by the device. Since your rom is reporting that you have NFC (but you dont have) then the app crashes, so I think is more an issue with the rom than with PHONK now.
If you can post a log I can try to add some try catches. I think that since your setup is very unlikely to happen to another person, dont you think?

marclava, 2020-07-14:
My phone is also a Xiaomi, model Mi A2 Lite (Redmi 6 Pro); it does not support NFC, and the Phonk example shows a proper alert ("This device does not support NFC"), very probably because my ROM (LineageOS) correctly detects that NFC is not supported.

victordiaz, 2020-07-14:
Thanks @marclava for checking it out

BillisC, 2020-07-14:
Well looks like a rom issue in the endm This should conclude the nfc problems

victordiaz, 2020-07-15:
@BillisC ok! thanks for your initial report, I made me help improving and fixing a bunch of NFC related stuff :)

closed 2020-07-15 by victordiaz