victordiaz/PHONK #37: Idea: autoselect enclosing function when evaluating a single line in the editor

hamoid, 2020-07-04:
Currently, when editing a Processing based program, I can press Ctrl+Shift+X to reevaluate the selection. 

This works if I select the whole `draw` function for example. 

But I don't select anything, it will give an error because evaluating `p.stroke(255, 0, 0)` out of any context fails (`p` is unknown).

The SuperCollider IDE solves this by implementing a feature that autoselects the enclosing function. That would very convenient for experimentation. In my example, I can keep changing the RGB values of stroke and press Ctrl+Shift+X to observe the results.

One approach would be: if nothing is selected, start scanning from the caret position backwards until `function` is found, then forwards until the matching `}` is found (accounting for `if` statements and `for` loops), then evaluate that block leaving the caret position untouched.

victordiaz, 2020-07-04:
I would love to see that!
I'm just wondering how would it work in a case where I want to execute only one line within some brackets (In this case I dont need the internal context)

The live execution works in more use cases than Processing, so how it work for more scenarios? Maybe different shortcuts?

By the way @hamoid , what is the default shortcut for other live coding environments?

hamoid, 2020-07-05:
I think shortcuts involving [enter] would be nice. Like ctrl+enter, shift+enter, alt+enter or ctrl+shift+enter... something like that.

This is what I have in SuperCollider, but I don't remember the defaults:
[image]

There could be two shortcuts: one for the current line, and one for the current selection or the surrounding function if there's no selection.