victordiaz/PHONK #40: Idea: close app with a command

LukasEdl, 2020-07-04:
**The Idea**  
A command to close phonk and not only the current script.

**What would be the benefit**

Im currently trying to use phonk as a way to send mqtt commands quickly.
My idea was to create a script which just connect to the mqtt broker, send the message and then closes the app. Then i create a shortcut to the desktop and i would be able to send mqtt commands with one button press. However, the current `app.close()` command only closes the script, but not the application and i end up on the script overview screen.  

With an command to close the application you could create buttons to controll your applications easy and fast.

victordiaz, 2020-07-04:
Hello @LukasEdl 
This is indeed a very good use! I think there might be two approaches for this:

1) app.close() closes the script if the script was launched from the main app, but if the script was launched from a shortcut, the app should close entirely without going to the main screen
2) Another method can be added, like closePhonk() / closeAll() / closeEverything() which is a more radical approach but maybe more effective for certain uses. 

Do you think the 1) option might work for you?

LukasEdl, 2020-07-05:
Hi,

option 1) actually works for me. However it only works if phonk is closed when i launch the script from a shortcut. When phonk is running in the background, it will show the script overview after the `app.close()` command was called.

victordiaz, 2020-07-05:
I see! Then it might be interesting to add a function to close the parent app for certain situations!

Right now PHONK works in the following way.

- PHONK app -> This is the main UI 
- PHONK service -> this runs in the background and contains the servers to use the Wed Editor
- PHONK Apprunner -> this is actually what runs your script

When you run your script, you tell the service to run the AppRunner with your script loaded.
The methods that you can use in the PHONK Apprunner do not affect the PHONK app or PHONK service. The reason is that eventually there is going to be a way to create an APK from your script, which will contain just the PHONK Apprunner and your script and assets.
But I think your use case is totally worth it and useful for lots of people so I can a way to have more control on what to close.

Maybe extending the `.close()` method as follows can give good flexibility

```
app.close({
  phonk_app: true,
  phonk_service: true,
  script: true
})
```

victordiaz, 2020-07-07:
I just added a new method `app.finish()` that closes the script and the parent app.
It will be included in this week update

LukasEdl, 2020-07-09:
nice, thanks for adding it

victordiaz, 2020-07-14:
Hi @LukasEdl 
The new version is online with the changes.
Feel free to close the issue if everything works fine for you :)

LukasEdl, 2020-07-15:
Hi @victordiaz it works for my case.

One thing, if Phonk is not open and i open my script which closes with app.finish() there is still the process in my process list.

victordiaz, 2020-07-15:
Hi @LukasEdl,
I think the process might be there since Android manages the lifecycle of apps. You might close it but Android might keep it for a bit in memory in case it gets reused. When Android needs resources it will properly kill the processes. "Hard killing" an app is not recommended in Android
Do you think it might be the case?

LukasEdl, 2020-07-16:
It could be that case, however i dont do alot of android development, so i cant tell you that.

The function however provides what i my idea was so i gonna close the ticket.

closed 2020-07-16 by LukasEdl