victordiaz/PHONK #42: Unable to download it via Play Store (incompatible device)

elnezah, 2020-07-05:
**Problem description**
I was not able to download/install Phonk to my device through Play Store, it says that my device is not compatible. I downloaded the APK from phonk.io and it works perfect (as far as I have tested it), so it looks as it is compatible really.

I have had similar issues with other (few) apps and I am quite sure it happens because my tablet has an Intel Atom processor.

**Device model**
Lenovo Yoga Book (YB1-X90F)

**Android version**
7.1.1

**Steps to reproduce the bug**
In the description.

victordiaz, 2020-07-06:
Is this model? If so, Google Play reports that is supported :/

[image]

victordiaz, 2020-08-06:
I close the issue since I could not replicate and figure out the problem

closed 2020-08-06 by victordiaz