victordiaz/PHONK #43: Any chance to develop the UI for landscape orientation?

elnezah, 2020-07-05:
**Problem description**
Not sure if this is already done and I have just not seen it: I use to use Phonk to develop right in my tablet using physical keyboard. This setting force me to use my tablet in landscape mode. Then when I run the program, the orientation of it is portrait, which is not convenient. Is there any change to tell Phonk I want to stay landscape, so it draws the UI I program landscape (x, y, text orientation, etc)?

Hint: the device information bubble tells me that Phonk recognizes my tablet is in landscape mode and also the embedded IDE is in perfectly comfortable landscape setup.

Hint: it is clear that Phonk knows the orientation of my device even during the run of my code, because if I turn it, the program will restart when going from landscape to portrait (or vice versa) - So the onPause-onResume cycle happens, even though the UI will not change at all.

**Device model**
Lenovo Yoga Book (YB1-X90F)

**Android version**
7.1.1

**Steps to reproduce the bug**
Use Phonk in landscape mode, run a program with UI. If you don't turn the Android device, the coordinates of your elements will be 0,0 in the bottom-left corner, the text will appear vertical (bottom to top).

victordiaz, 2020-07-06:
Hi @elnezah 

There is a way to launch a script in landscape mode. Please check Advanced -> App Config example.
Inside there is a app.conf file where you can specify the orientation to `"orientation": "landscape"`

Do you think having an example inside the Graphical User Interface folder will make it more easy to discover?

victordiaz, 2020-08-07:
Hi @elnezah 
In the las release 1.2.9 PHONK launches scripts in the same orientation as it was previously, so it should improve the experience

victordiaz, 2020-08-10:
I close this issue since it is resolved

closed 2020-08-10 by victordiaz