victordiaz/PHONK #45: Is there any problem with Camera.turnOnFlash(true)?

nikunjbhatt, 2020-07-06:
I am exploring this awesome PHONK project.

In the main.js file, if I write only one line `Camera.turnOnFlash(true)` and run the PHONK project, then should it turn the flash on? I tried this after looking in the Reference guide, but it is not working. I also tried below lines:
`camera.turnOnFlash(true)`
`device.Camera.turnOnFlash(true)`
`device.camera.turnOnFlash(true)`
But none of these are working. This is the error showing in the Console in the PHONK editor in web browser:
`ReferenceError: "Camera" is not defined. (test proj#1)`

Am I missing anything? Am I doing any mistake? Or is there any bug?

I tried these in my phone Moto X Play (XT1562) with Android 7.1.1 (Nougat).

victordiaz, 2020-07-06:
Hello @nikunjbhatt 

Please check the Media -> Camera example.
The current implementation needs the camera to be active in order to turn on the flash, since it uses an old Android implementation that needed the camera to be active.

I'm aware its quite possible to turn on the flash without the camera, I will add that to the TODO list and include it for the next release.

Maybe the camera example can work for now :)
Thanks for the report

nikunjbhatt, 2020-07-06:
Ok. Got it. Thanks for quick reply.

nikunjbhatt, 2020-07-06:
I don't have much idea about Android development, but as per you said that old Android implementation needed active to work with flash, which means new Android implementation doesn't require active camera then I think the API should be changed to `device.turnOnFlash(boolean)` instead of `camera.turnOnFlash(boolean)`. Also, I suggest to remove the word `On` from the function name, or even remove the words `turnOn`, to make it just `device.flash(boolean)`.

victordiaz, 2020-07-07:
That's a good idea!
On the other side, there is some devices that have multiple flash lights (front and rear), so I think is good that the camera has its own method to turn on the flash light.
I will check how is turning the flash without the Camera and include it. It makes sense to include it inside the device object.
I like to have an action verb for some methods, so maybe `device.activateFlash(boolean)` might sound ok?

nikunjbhatt, 2020-07-11:
@victordiaz 
Sorry for very late replying, I had completely forgot.

I don't think using verb in methods which are doing toggling things is good. Or there should be 2 different methods for 2 different states, for example: `device.activateFlash()` and `device.deactivateFlash()`.

victordiaz, 2020-07-11:
@nikunjbhatt I agree It's better without the verb. Thanks for the suggestion :)

victordiaz, 2020-08-06:
@nikunjbhatt 

You can now turn on/off the flash light using 
`device.flashLight(boolean)`

Feel free to check it and close the issue if it's the expected behaviour. Thanks!

nikunjbhatt, 2020-08-06:
@victordiaz 
Thanks. It's working!

By the way, in the README.me of the PHONK repo, you have told to contact you by email but I can't find your email anywhere, not in the README.md file, not in your GitHub profile, not on your personal website, not on the phonk.app website!

May I know how the reference documentation is generated? In the reference documentation, it is showing as `String` for all parameters for ui.addToggle() function (and for many other functions), but actually the 'text' param is an Array (of String) and the other params, w h x y, are of type object. So, how to show them correctly in the reference documentation? Is it a problem of the documentation generator or improper documentation in the code files?

closed 2020-08-06 by nikunjbhatt

victordiaz, 2020-08-06:
Hi @nikunjbhatt 

Feel free to join this discord server so we can chat [https://discord.gg/Rt2mkWp](https://discord.gg/Rt2mkWp)
I'm sorry if there is things missing everywhere, I lack a bit of time to put everything nicely :)

Regarding the documentation. Is it generated with and older version of Java and I had a mess in my system since it updated Java and now the generator does not want to run :/ so the docs are generated some time ago and many methods have changed since then.

It would be nice to dockerize the generator so I can run it easily before a new release and things can be up-to-date