victordiaz/PHONK #46: Util.parseBytes only supports chars

txf-, 2020-07-07:
```
    public Object parseBytes(byte[] bytes, String type) {
        BluetoothBytesParser parser = new BluetoothBytesParser(bytes);

        switch (type) {
            case "uint8":
                return parser.getIntValue(BluetoothBytesParser.FORMAT_UINT8);
            case "string":
                return parser.getStringValue(0);
            default:
                return null;
        }
    }
```
Ideally this section should support more datatypes than just 8 bit values. Presumably BluetoothBytesParser supports more data types?

victordiaz, 2020-07-07:
@txf- 
It seems to support float, int and longs as far as a I can see. I will include them and release it in the next version.
Thanks for commenting on this

victordiaz, 2020-07-11:
I included some more data types although I only managed to test uint8 and strings. For some reason float does not work for me but maybe I'm just sending wrongly the data from my ESP

victordiaz, 2020-07-14:
Hi @txf- 
The new version is online with the changes.
Let me know if that works as intented

Thanks

victordiaz, 2020-08-01:
I close the issue since I think a previous commit improved the situation :)

closed 2020-08-01 by victordiaz