victordiaz/PHONK #47: No Permission to send SMS, smsSend()

Nic7C5, 2020-07-08:
*Try to add information such as (if possible)*

No Permission to send SMS

Blackberry Key 2 (BBF100-6)

Android 8.1.0 (Kernel 4.4.78...) Security Patch May 2020

calling the function

`smsSend(012345,'test')`

throws the error `Wrapped java.lang.SecurityExeption: Sending SMS message: uid 10238 does not have andoird.permissionSEND_SMS.`
 [](https://support.google.com/googleplay/android-developer/answer/9047303?hl=en)

victordiaz, 2020-07-08:
Hello @Nic7C5 

Google Play does not allow anymore apps that access the SMS, so for that reason I created a separated extended APK that you have to download from here [https://github.com/victordiaz/PHONK/releases/tag/1.2.5](https://github.com/victordiaz/PHONK/releases/tag/1.2.5)

Can you try to install that version and let me know if it works? (you have to uninstall the previous version)

Nic7C5, 2020-07-08:
I just updated my first post. At first I used version 1.2.5 from the Playstore. Now I installed the 1.2.5 extended apk from here. The error is still the same.

victordiaz, 2020-07-08:
Ok I see, I'll have a look later with more detail and see what's going on :)
Thanks for reporting

victordiaz, 2020-07-08:
Hello @Nic7C5 

Could you check to install `phonk-1.2.6-alpha-4-extended-release.apk` from [https://github.com/victordiaz/PHONK/releases/tag/1.2.6-alpha](https://github.com/victordiaz/PHONK/releases/tag/1.2.6-alpha) and tell me if it works for you?

Thanks

victordiaz, 2020-07-08:
@Nic7C5 
One thing I saw in your first comment the method `device.smsSend(012345 , 'test')` should be instead `device. smsSend('012345', 'test')` 
The number is a string since you can write Phone numbers with the + prefix

Nic7C5, 2020-07-08:
It's working with the alpha 4 (and number as string). Thanks!

victordiaz, 2020-07-08:
I will keep the issue open until the next release. Thanks for testing it :)

victordiaz, 2020-07-14:
The new version is online with the changes.
Feel free to close the issue if everything works fine for you :)

closed 2020-07-17 by Nic7C5

adasq, 2020-08-04:
When can we expect this version to be published into Google Play Store?

victordiaz, 2020-08-04:
Hi @adasq 
Google Play is nowadays very strict with some features like sending SMS, so if you are not a big company or your SMS sending is justified somehow they don't allow it.
For that reason there is an Extended version [https://github.com/victordiaz/PHONK/releases](https://github.com/victordiaz/PHONK/releases) here that you can download and install in your phone or tablet that has SMS functionality.

I hope it works for you! Let me know if it does not.

adasq, 2020-08-04:
Hey @victordiaz,

Thanks for the response. I tried the released APK, but unfortunately, it crashes after downloading examples.

victordiaz, 2020-08-04:
@adasq try to uninstall all the PHONK versions in your phone and then reinstall the extended version. Let me know if that works!

adasq, 2020-08-04:
Well, it works!
Really appreciate your work. Thank you.