victordiaz/PHONK #49: LiveCodingOverlay has small font size

victordiaz, 2020-07-08:
**Problem description**
As reported in #32 
example Advanced / LiveCoding : when pressing Ctrl+Shift+X from the Web Editor, there is a very tiny unreadable line of text displayed temporarily at the bottom of the application on the tablet.

**Device model**
Acer Iconia B1-A71 tablet

**Android version**
Android 4.1.2. 

**Steps to reproduce the bug**

SuperUserNameMan, 2020-07-15:
I've just found the tiny text that appears at the bottom of the tablet (when pressing Ctrl + Return) is a copy of the code-line that is updated.

It is still very tiny and unreadable with 1.2.7

victordiaz, 2020-07-15:
Hi, I forgot to update the example :) I just changed it. 

In the code itself you can change the size using
``` .textSize(12)```

I will include the change in the next release
Thanks :)

victordiaz, 2020-07-15:
By the way, the liveCoding thing is just a way to show people what line is executed in each time. It's a bit geeky buy found it useful for demonstration purposes :)

victordiaz, 2020-08-05:
@SuperUserNameMan The example is adjusted in 1.2.8 so the text looks a bit bigger. Feel free to close it after you try it :)

SuperUserNameMan, 2020-08-06:
@victordiaz : yes it's better.

> Feel free to close it after you try it :)

PS : i can't close issues if that what you meant.

victordiaz, 2020-08-06:
Thx, for checking. I realized I opened the issue :facepalm:  I close it :smile:

closed 2020-08-06 by victordiaz