victordiaz/PHONK #50: app console has small font size

victordiaz, 2020-07-08:
**Problem description**
As reported in #32 
The text into the kind of error box is very small an almost unreadable (but i've just noticed they appear into the Console box of the WebEditor too).

**Device model**
Acer Iconia B1-A71 tablet

**Android version**
Android 4.1.2. 

**Steps to reproduce the bug**

victordiaz, 2020-07-11:
@SuperUserNameMan could you send me an screenshot?

SuperUserNameMan, 2020-07-12:
Note : The Display / Font Size parameter of my tablet is set to "Normal". 
The actual width of the screen is 9cm and its diagonal is 7 inches.

[image]

victordiaz, 2020-07-15:
@SuperUserNameMan I increased tiny bit the default console size. Do you think is enough or should I have a bigger text size for tablets?

SuperUserNameMan, 2020-07-15:
Oh I forgot to keep you updated with this change when testing new version yesterday. sorry.

Yes, now it has same size as the text at the top. It is readable.

You can close this issue :-)

victordiaz, 2020-07-15:
Perfect, thx :)

closed 2020-07-15 by victordiaz