victordiaz/PHONK #52: Processing / LibPd crashes on Android 4.1

victordiaz, 2020-07-08:
**Problem description**
As reported in #32 
Processing and LibPd crash on Android 4.1

**Device model**
Acer Iconia B1-A71 tablet

**Android version**
Android 4.1.2. 

**Steps to reproduce the bug**

SuperUserNameMan, 2020-07-09:
Little precision : It crashes with 1.2.5, but it worked with 1.0.0.

garydion, 2020-07-09:
Can also confirm the Processing example in 1.2.5 crashes Phonk in 4.4.4.

victordiaz, 2020-07-09:
@garydion Thanks for letting me know
Do you have any log?

PHONK 1.0.0 used an older version of Processing core, maybe there is something in newer versions that breaks in old models.
I'll have a look into this asap

garydion, 2020-07-09:
I'll check for any log messages tonight.

SuperUserNameMan, 2020-07-09:
> PHONK 1.0.0 used an older version of Processing core, maybe there is something in newer versions that breaks in old models.

Sorry, my little precision was missing an important precision : 
- Only LibPD works with 1.0.0. 
- Processing crashes with both 1.2.5 and 1.0.0.

victordiaz, 2020-07-11:
[According to this](https://github.com/Calsign/APDE/commit/bb0ed8b5268843811885957d0f77ff598e8b36d1) it seems that Processing is trying to look for some Wearable classes that only exists in higher Android versions. 

I'm going to fork Processing core and check if I can "easily" remove those dependencies. If not I will revert the library to the old one. I prefer to have the widest support across versions rather than newer features.