victordiaz/PHONK #53: OpenStreetMaps do not work on Android 4.1

victordiaz, 2020-07-08:
**Problem description**
As mentioned in #32 
example GUI / Maps (OpenStreetMaps) : It does not display any map. The background is a gray grid with a "MAPS (OPENSTREEMAPS)" title in bold yellow, a red circle, a cyan circle, two blue segments and a pinpoint-location icone. Zooming and scrolling moves them randomly.

**Device model**
Acer Iconia B1-A71

**Android version**
4.1.2

**Steps to reproduce the bug**

victordiaz, 2020-07-11:
I had a look on other Android 4.x.x devices and it seems there is a problem with the OSM library PHONK is using.
I opened an issued here and see if there is some answer. [https://github.com/osmdroid/osmdroid/issues/1617](https://github.com/osmdroid/osmdroid/issues/1617)

victordiaz, 2020-08-05:
@SuperUserNameMan

victordiaz, 2020-08-05:
@SuperUserNameMan In the new version 1.2.8 maps should work correctly on 4.x.x device. Could you try it out? Thanks!

SuperUserNameMan, 2020-08-06:
@victordiaz : Yes, I confirm it works ! :-D

victordiaz, 2020-08-06:
Ok! thx for checking @SuperUserNameMan ! I close the issue then :smile:

commit 30f5ea9a (2020-08-01): Update in osmdroid fixes #53
closed 2020-08-06 by victordiaz