victordiaz/PHONK #54: Firebase integration

gschora, 2020-07-10:
Would it be possible to integrate Firebase?

This would be great to log some data into a database or use the ML kit since you've already integrated the camera...

And you really should setup a donation button somewhere, because you have no clue how long I've searched for something like this :-)

victordiaz, 2020-07-10:
Hi @gschora 

1) I think is very interesting to add Firebase. My only concern is that some people want to have a de-googlified version of PHONK but I would not mind to add a variant with non-free stuff, or a way to add libraries on demand ( I dont know how to do this yet in android 😅)
I will create a new branch and try to add it during the next days.

2) Thanks for your comment, is flattering :) 
I made recently a Patreon 
[https://www.patreon.com/victornomad](https://www.patreon.com/victornomad) and a [https://www.buymeacoffee.com/victordiaz](https://www.buymeacoffee.com/victordiaz) but forgot to put the links 😅
Thanks for reminding me.

gschora, 2020-07-10:
I don't know if this would be a problem for people, since you are already using Google services in your text recognition example.
BUT to have at least the option would be awesome, and realtime data exchange between devices would be so much easier...

and thanks for the link :-)

victordiaz, 2020-07-10:
Hi @gschora :)
The text recognition example uses the "recognition service" that it's used system wide. For some phones will be Google for some others not (ie: roms, Huawei, devices in China, some samsungs, etc) :) 
PHONK does not use Google Play services library which I think is a good decision to allow people use it without restrictions.

Nevertheless I think Firebase might be superconvinient so I'm happy to include it. I just have to check how to do it in the right way :)

marclava, 2020-07-11:
Firebase is a service with a REST API that *requires* Google Play services for authentication. I would suggest to ensure that with added Firebase support, Phonk would still work on ungoogled phones (if possible).