victordiaz/PHONK #57: Example Media / Camera crashs phonk 1.2.x on Android 4.1.2

SuperUserNameMan, 2020-07-14:
Hello again,

I'm testing the new Phonk 1.2.7 on my Acer B7-A71 tablet (Android 4.1.2).

**Problem description**
PHONK 1.2.x crashes when running example media/camera.

**Device model**
Acer B7-A71 tablet 

CPU : MT6517 armv7l
GPU : PowerVR SGX 531
RAM : 512 MB
Display : 600x1024 @ 169 dpi

**Android version**
Android 4.1.2
API Level 16
OpenGL ES 2.0

victordiaz, 2020-07-14:
Hello @SuperUserNameMan 

1) Did it work before?
2) Did you update the examples
3) Do you get any error?

Thanks

SuperUserNameMan, 2020-07-14:
No it never worked on this tablet. 
I thought I told you already, but I might have forgot.

SuperUserNameMan, 2020-07-14:
Oh wait ! ``useCamera('front')`` instead of `back` works this time.

It did not work at all with previous version. 

So it only crash when ``useCamera('back')``.

victordiaz, 2020-07-14:
Thanks @SuperUserNameMan 
I will add a camara availabilty check before you can use it :)

SuperUserNameMan, 2020-08-06:
@victordiaz : I've tested this example again with the new 1.2.8. And it does not work anymore. It displays this error :
``[ :-( problem! ]  Wrapped java.lang.NullPointerException (Camera#12)``

victordiaz, 2020-08-06:
@SuperUserNameMan What is your expected behavior? I will have a deeper look these days

SuperUserNameMan, 2020-08-06:
> @SuperUserNameMan What is your expected behavior? I will have a deeper look these days

Oh sorry, nevermind ... my bad. I tested again and it works. 
I must have forgot to save the script before running it again into the webeditor after i replaced "back" with "front" (as my tablet does not have back camera).

Though, what do you think of this behavior for ``media.useCamera()`` :
- if the requested camera is not available, automatically try with the other one ?
````
    public PCamera useCamera(String camera) {
        PCamera pCamera = new PCamera(getAppRunner(), camera);
        if ( pCamera == NULL ) pCamera = new PCamera( getAppRunner(), camera == "front" ? "back" : "front" );
        return pCamera;
    }
````

closed 2020-08-06 by SuperUserNameMan