victordiaz/PHONK #59: example Advanced / Background App displays "script has crashed"

SuperUserNameMan, 2020-07-15:
Hello again !

When I run the Background App example, the background-app show up into my notification list.

And when I click/touch it into the notification list, the screen turn black and a tiny box at the bottom says "The script has crashed :(" and the PHonk application seems to reboot.

**Device model**
Acer Iconia B1-A71 tablet with Android 4.1.2

**Steps to reproduce the bug**

Run the Background App example, open the notification list, click/touch the "Background app" notification.

victordiaz, 2020-07-15:
Thx for reporting this. Background services is something I want to fix, so I will take this issue as an opportunity to improve it :)
Basically the Background service runs scripts in the background in cases where you dont need a UI or you don't want the script to keep running when the screen is off.
ie: music player or communicating with a server.

SuperUserNameMan, 2020-08-08:
Oh this one is solved BTW !

closed 2020-08-08 by SuperUserNameMan