victordiaz/PHONK #60: Code changes do not take effect until PHONK is restarted.

briantilley, 2020-07-30:
Code edits I make to my project on both the mobile text editor and the web editor do not take effect the next time I save and run my project. As a workaround, I have to do to following:

Save my changes
Stop the webserver
Remove the app from recent apps ("kill it")
Re-open PHONK
Start my project (works from web editor and phone)

I am using a OnePlus 5 (version w/ 6GB ram, 64GB storage) on Android 10 (Oxygen OS 10.0.0)

To reproduce, open the project, edit in web browser, and run the project. Stop, save new changes, then run again, and old behavior persists.

victordiaz, 2020-07-31:
Hi @briantilley 

I cannot reproduce the bug with any of my devices or emulators :/
When you save the code from the Web Editor, do you get the notification that is "saved"?

Can you double check if you gave permissions to access the device storage?

briantilley, 2020-07-31:
The "Saved" notification does go through successfully (changes to yellow) and I can verify that by looking at the updated code from the app. I just checked, PHONK has storage access permission. Only one version is installed, from the Google Play store. If it would help, I can provide the project files later today.

victordiaz, 2020-07-31:
There is a discord channel here https://discord.gg/Rt2mkWp

Feel free to enter if you think we can solve the problem faster :)