victordiaz/PHONK #66: Knob `decimals()` method has no effect

Jakoma02, 2020-08-05:
It seems to me that the `decimals()` method of a knob has no effect (as opposed to changing the number of displayed decimal places). The bug can be easily reproduced by using the method with any argument other than 2.

I believe that the problem lies in the `PKnob` java class, where the `formatString` is only used in the **constructor** to create a `DecimalFormat` instance `df` that is later used when drawing the knob. As the `decimals()` method only changes `formatString`, `df` never gets updated and the change has no effect.

Sorry if the mistake is on my side, I could have easily overlooked something.

victordiaz, 2020-08-05:
Hello @Jakoma02 
You are totally right, thanks for letting me know. I'm fixing it ASAP and include it in tonight's release :)

victordiaz, 2020-08-05:
@Jakoma02 the new version is uploaded and should be online

victordiaz, 2020-08-10:
I close this issue since it is resolved

commit c5c6e063 (2020-08-05): decimal method not working fixed #66
closed 2020-08-10 by victordiaz