victordiaz/PHONK #68: Background application example can't be stopped on Android 9 / MIUI 11

SuperUserNameMan, 2020-08-06:
Hello,

When I test the "backgound application" example on Android 4.1.2 and Android 5.1, the background app appears into the notification list and can be stopped with a "stop button".

But when I run it on my Xiaomi Redmi 7 with MIUI11 (Xiaomi customized version of Android 9), it does not appear into the notification list and I can't stop it unless I close Phonk.

I don't know if it is related to Android 9 or to MIUI11.

**Problem description**

Can't close the "background application" on Android 9 / MIUI 11, because it does not appear into the notification list.

**Device model**

Xiaomi Redmi 7

**Android version**

MIUI 11 (customized version of Android 9)

**Steps to reproduce the bug**

Run the example, and look into the notification list where it should appear.

victordiaz, 2020-08-07:
Hi @SuperUserNameMan 
The version 1.2.9 should fix the problem. Let me know if it works for you

SuperUserNameMan, 2020-08-08:
Hello @victordiaz : yes it works ! :-)

[image]

closed 2020-08-08 by SuperUserNameMan