victordiaz/PHONK #69: Add better support for Android TV ?

SuperUserNameMan, 2020-08-06:
Hello again,

I don't know if Android TV are part of your goal, but I was curious to test Phonk on my Android TV 5.1.

I could not try all examples, but here is a list of some issues I've found so far :

- [ ] Phonk icon does not appear into the "launcher" or "desktop" (I have to go deeper into the application management menu to find and run Phonk)
- [ ] Remote control navigation into the Phonk user interface is not well supported : I can manage to open the Phonk menu and navigate into the "New","Settings","Help" and "About" submenu and scroll the pages, but I could not navigate into the Examples folders.
- [ ] I confirm the remote control buttons emit KeyEvents, but remote control navigation did not work at all with the GUI examples.
- [ ] Phonk run in landscape mode, but almost all example run in portrait by default (which is not very "neck-friendly" on a TV ;-P ) : maybe Phonk could detect when running on an Android TV and force landscape mode ?
- [ ] The camera example displays 2 errors into the web editor : ``Wrapped java.lang.IllegalArgumentException: Cannot add a null child view to a ViewGroup (Camera#12)`` (note : the TV does not have camera)
- [ ] The LibPD example could not work because the TV does not have accelerator (but I confirm LibPD works on it)

If you want me to do more tests on this device, or provide more information, don't be afraid to ask me ! :-)

SuperUserNameMan, 2020-08-06:
device

    model name: QM152E
    type: tablet
    manufacturer: PHILIPS

screen

    screen resolution: 1280 x 720
    screen dpi: 213
    screen: true
    brightness: 102
    orientation: landscape

script

    running script: none

other

    total memory: 28966823
    used memory: 27155799
    battery level: 100

victordiaz, 2020-08-06:
Hi @SuperUserNameMan 
Thanks for the report! I can change a few things I think to make it a bit more compatible although I don't have any Android TV!

victordiaz, 2020-08-07:
I modified a bit PHONK so now the script will launch with the same orientation as it was previously. It should improve the 4th point.
I think for the LibPd example I can modify it so it wont need the accelerometer :)

SuperUserNameMan, 2020-08-08:
@victordiaz : 
> I modified a bit PHONK so now the script will launch with the same orientation as it was previously

yes it works :-) 
Only the Processing example keeps displaying in portrait.

> I can change a few things I think to make it a bit more compatible although I don't have any Android TV!

I don't know if there will be a lot of Android TV owners among the present and future PHONK! community. But I personally find it quite cool to know these deprecating Android TV will keep the potential to easily become a computer for a young (or older) programmer thanks to PHONK.

Regarding Android TVs or smart TV in general, that's not something i would not recommend to buy to anyone, because it deprecates too fast. (I had to hack it recently to make youtube work again).

When I will have more free time, I will give a look at PHONK source code and see how I could contribute more effectively to help you extend compatibility to Android TVs.

:-)