victordiaz/PHONK #70: camera.takePicture

42loop, 2020-08-06:
phonk 1.2.8: when running the camera example, it shows the image,
but when trying to take a picture:

Can't find method io.phonk.runner.apprunner.api.media.PCamera.takePicture(string,function). (Camera#16)

while the flash function works

if I would understand how it all works, i'd like to contribute - i see the declaration in the java files but still phonk can't find the method ???

victordiaz, 2020-08-06:
Hi @42loop 

Feel free to join this Discord channel [https://discord.gg/Rt2mkWp](https://discord.gg/Rt2mkWp) so I can explain how things work and maybe document everything along the way :)

victordiaz, 2020-08-07:
Hi @42loop 
The version 1.2.9 should has an updated example + some fixes so the camera should work now