victordiaz/PHONK #72: Customizable shortcut image

LukasEdl, 2020-08-07:
**Is your feature request related to a problem? Please describe.**
I use PHONK for controlling devices connected over mqtt. I create a shortcut which calls a script that just connects to the mqtt broker and publishes a message. Currently the only differentiation is the text of the shortcut.

**Describe the solution you'd like**
When you create the shortcut or during the writing of the script, you can select an image for the shortcut. The image could be just an image from the device or from a image collection provided by PHNOK

victordiaz, 2020-08-07:
Hi @LukasEdl 
I just pushed a new version 1.2.9 with the possibility to add a custom icon.
You just have to include a `icon.png` inside the project and it should work. 
In `examples/Graphical User Interface` the example `Styling & Theming` has an icon as a way to show how

LukasEdl, 2020-08-08:
Hi @victordiaz,
it works like a charm, thank you for implementing it.

I was thinking a little bit and was wondering if it would be even possible to change the icon within the script?

victordiaz, 2020-08-08:
@LukasEdl nice to read :)
Do you mean changing the shortcut icon from the script? the icon in the list?

LukasEdl, 2020-08-08:
I mean the icon of the shortcut

victordiaz, 2020-08-10:
Hi @LukasEdl 
I was reading about it and it seems that you cannot change the icon with the official Android APIs, there are some tricks that works in certain android versions / launchers, so I think it would be a bit of headache to implement something that works half-way.

I'm thinking though that a solution would be to create a Launcher Widget that can list some desired scripts within. I think having  the scripts it in a widget we can have more control and not only change the icon but also some text, info, etc

LukasEdl, 2020-08-13:
Hi @victordiaz 
The use of Widgets would be awsome, since it would be perfect for my use case and probably usefull for other use cases. 

I'm gonna close the ticket since its original feature request was implemented.

closed 2020-08-13 by LukasEdl