victordiaz/PHONK #74: network.wifiScan() does not work with Android 9 / MIUI 11

SuperUserNameMan, 2020-08-08:
Hello again,

I tried the WifiScan example on my Xiaomi Redmi 7 (android 9 / MIUI 11 ), and it does not displays the list of network.

Some more tests shows that the ``network.wifiScan()`` function does not call the callback on this device.

I checked the permissions allowed to Phink! and they seems to be enabled for accessing the list of wifi network.

**Problem description**

network.wifiScan() does not call the callback.

**Device model**

Xiaomi Redmi 7 

**Android version**

MIUI 11 (customized version of Android 9 )

**Steps to reproduce the bug**

Run the WifiScan example on my phone ?

victordiaz, 2020-08-10:
Hi @SuperUserNameMan 
Could you check in another device? Does it give any error?

SuperUserNameMan, 2020-08-10:
Hello @victordiaz :

> Does it give any error?

I get no error with this device.

> Could you check in another device?

It works with my Android 4.1.2 tablet, and with my Android TV 5.1.1 tv.