victordiaz/PHONK #75: ui.addTextList() : some lines are displayed in very small

SuperUserNameMan, 2020-08-08:
Hello again again,

By testing the WifiScan example on my Android 4.1.2 tablet, and on my Android TV 5.1.1, I've found out that some lines of the list of networks were displayed in very very small (unreadable).

After some more tests, it seems that it comes from ``ui.addTextList().add()``.

**Problem description**

Some lines of text in the ``ui.addTextList()`` list are extremely small and unreadable.

**Device model** & **Android version**

4.1.2 (tablet) and 5.1.1 (TV)

**Steps to reproduce the bug**

Run the WifiScan example and my devices ?

SuperUserNameMan, 2020-08-08:
[image]

victordiaz, 2020-08-28:
mm, strange! :/ I'll have a look