victordiaz/PHONK #76: Missing ; after statement

EuphoricPenguin, 2020-08-13:
`let a = 2;` results in a ASI error where the code won't run, meaning variables are effectively useless. This happens with `var` as well, and seems to happen from a numerated "run" of the code. I'm running a Pixel 4 on the latest stable release of Android 10. I installed the latest extended release of PHONK, and ran into this issue.

victordiaz, 2020-08-13:
Hi @EuphoricPenguin 

let is not supported since the Rhino Javascript interpreter does run modern jacascript (ie ES6).
With var should works perfectly.
What is the issue that you are facing with var?

EuphoricPenguin, 2020-08-13:
Identical issue.

victordiaz, 2020-08-13:
There is this issue describing the problem with the current interpreter.

https://github.com/victordiaz/PHONK/issues/4

Feel free to join the discussion there :)