victordiaz/PHONK #78: Tasker integration

thejeshgn, 2020-08-26:
Add ability to send and receit data from Tasker. Phonk can look like a plugin for Tasker.

victordiaz, 2020-08-26:
Hi @thejeshgn 
Just out of curiosity, what uses cases can you see?

thejeshgn, 2020-09-06:
I have couple of scenes in Tasker. In one of them I wanted to trigger a script in Phonk

victordiaz, 2020-10-05:
Hi @thejeshgn 

I remembered we can launch PHONK scripts using an intent.
For example in ADB we can do something like 

```
adb shell am start -n "io.phonk.extended/io.phonk.runner.AppRunnerActivity" -e "projectFolder" "playground/User\ Projects" -e "projectName" "MyProject"
```

To launch the script "My Project" inside the "User Projects" folder. Have in mind that I'm using the "extended" version, if you use the normal one you should write `io.phonk` rather than `io.phonk.extended`.

If you want to translate this to Tasker, you can use the Send Intent action as follows

[image]

I hope this can help you out!