victordiaz/PHONK #81: Add headers to http requests

ae3e, 2020-09-30:
I have a web request with basic authentication so I need to had `Authorization : ` header

victordiaz, 2020-10-01:
Hi @ae3e 
Im happy to add the option to add the headers.
Could you help me providing a backend code or site where I could test it against?

ae3e, 2020-10-02:
Thanks for your quick answer.
You could use [Posman-echo](https://postman-echo.com) to test any web requests.
More specifically, you can test [headers](https://docs.postman-echo.com/#5d3595b3-5e8e-9e33-05ed-855c77298e4e) and [basic authentication](https://docs.postman-echo.com/#42c867ca-e72b-3307-169b-26a478b00641).

victordiaz, 2020-10-04:
Thx for pointing me to that service. Superuseful!
I had a look already and it's working. 

I'll upload tonight a testing version with a more generic HTTP request function :)

ae3e, 2020-10-05:
Thanks a lot!
I never compiled PHONK by myself so I'll wait for your next release to test it (unless it takes too long... i'll try compiled it! 😉)

victordiaz, 2020-10-05:
Hi @ae3e 
I uploaded a release here https://github.com/victordiaz/PHONK/releases/tag/1.3.1
You can try the following code

``` js
// the concatenation needs a .toString() there is a strange behaviour otherwise
var auth = ('Basic ' + util.stringToBase64('postman:password')).toString()

var req = network.httpRequest({
  method: 'GET',
  url: 'https://postman-echo.com/basic-auth',
  headers: {
    Authorization: auth
  },
}).onResponse(function (e) {
  console.log(e)
})
```

Let me know if does work for you :)

ae3e, 2020-10-06:
Thanks,
I just tested it with some home automation devices that need authentication and it works perfectly!
You're really doing a great work! 👍

closed 2020-10-06 by ae3e