victordiaz/PHONK #82: Mqtt: Enhanced error logs

LukasEdl, 2020-10-04:
**Is your feature request related to a problem? Please describe.**
I am currently working with the mqtt client and I ran, where i can't connect to a mqtt server.  I wanted to display the problem, however if i use the provided `onError` function, I only get an object like `{ "status": "error" }`. 

**Describe the solution you'd like**
It would be nice, if the error message would contain more information about the occurred error, maybe the error message or even the complete error, which was thrown by the paho-mqtt client.

victordiaz, 2020-10-05:
Hi @LukasEdl 
I created a new release with the changes. https://github.com/victordiaz/PHONK/releases/tag/1.3.1

Could you check if that's what you expected? Thx!

LukasEdl, 2020-10-06:
Hi @victordiaz,
yes this was exactly what i wanted, thanks for adding it.

closed 2020-10-06 by LukasEdl