victordiaz/PHONK #84: Compiling external libraries for LibPd?

jacksonmgoode, 2020-10-14:
Hi there,

Fantastic project - really nothing like it at the moment for getting Pure Data patches on a mobile device with ease (no OSC pass-thru some other app).

I was curious, however, if it would be possible to build PHONK with additional externals for Pd? I know it's possible through LibPd, ("EXTRA=true" as a Makefile option). It would be really interesting to see just how difficult this would be. If you have some idea of the options you used to build LibPd maybe I could test out adding new externals to PHONK?

victordiaz, 2020-10-14:
Hi @jacksonmgoode 

I would be happy to include more externals with PHONK.

At the moment, I'm using the libpd hosted in the maven repo as stated in this gradle file `https://github.com/victordiaz/PHONK/blob/master/phonk_apprunner/build.gradle`

Do you know if I have to compile from source to include externals? or can externals maybe be loaded on demand?

jacksonmgoode, 2020-10-14:
I believe they can be loaded on demand. I've been working on a [Bela](https://bela.io/) which is basically a glorified single-board computer - but adding externals is possible by simply compiling for the ARMv7 architecture and then dragging them into a specified folder.

I imagine that compiling the externals would be something an advanced user could manage - it's just a matter of uploading them somewhere within PHONK so that libpd can read them.

victordiaz, 2020-10-14:
I will have a look on that possibility.
Do you know a bit of Android? Could you set up an Android LibPd environment to compile a couple of externals so we can try?

jacksonmgoode, 2020-10-21:
I've only had some experience compiling for an Arduino (Bela), I'm not sure what the process would be like for Android - or what the architecture is like. Is there a tutorial somewhere I could follow?