victordiaz/PHONK #85: Can't create more than one shortcut to home screen

ae3e, 2020-11-03:
I have multiple scripts in PHONK app.
I can create one shortcut from any script to home screen. If I try to create a new shortcut, nothing happens. If I delete the existing shortcut then I'm able to create a new one from any script.

Is it not possible to create multiple shortcuts from different scripts?

I didn't find anything in logs.
Using PHONK extended and Android 5.1.1

victordiaz, 2020-11-06:
Hi @ae3e 
It seems that with newer versions is possible. I tried with Android 10 and you can create more than 1 shortcut. 
I will check an older version during the weekend and see why it does not work.

Thanks for reporting it!

victordiaz, 2020-11-16:
Hi @ae3e,
I just tried this in an emulator with Android 5.1 and I can create as many shortcuts as wanted without issues
A couple of questions to investigate this

1) are you using a custom launcher? 
2) do you use stock android?
3) can you try to add a script with a strange name? :) 
4) What is your device model?

Thanks!

ae3e, 2020-11-17:
Hi,

Thanks for your answer. Sorry to see that is related to my device config...
1. No
2. No
3. Tried some strange names (ie $45-ed#fR56) : scripts are created and ran but still can't create multiple shortcuts
4. [Wiko pulp fab 4G](https://world.wikomobile.com/m902-pulp-fab-4g) with factory defaults (Cheap and not common phone, I guess 😕 )

I never used Android Studio but I can try some debugging stuff with my phone if needed.

When trying to create shortcuts, I got this new log file :
```java
java.lang.RuntimeException: Unable to destroy activity {io.phonk.extended/io.phonk.MainActivity}: java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.ComponentName android.content.Intent.getComponent()' on a null object reference
	at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:3706)
	at android.app.ActivityThread.handleDestroyActivity(ActivityThread.java:3724)
	at android.app.ActivityThread.access$1400(ActivityThread.java:151)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1357)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:135)
	at android.app.ActivityThread.main(ActivityThread.java:5254)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:902)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:697)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'android.content.ComponentName android.content.Intent.getComponent()' on a null object reference
	at android.app.ContextImpl.validateServiceIntent(ContextImpl.java:1685)
	at android.app.ContextImpl.stopServiceCommon(ContextImpl.java:1745)
	at android.app.ContextImpl.stopService(ContextImpl.java:1706)
	at android.content.ContextWrapper.stopService(ContextWrapper.java:521)
	at io.phonk.MainActivity.stopServers(MainActivity.java:447)
	at io.phonk.MainActivity.onDestroy(MainActivity.java:204)
	at android.app.Activity.performDestroy(Activity.java:6212)
	at android.app.Instrumentation.callActivityOnDestroy(Instrumentation.java:1141)
	at android.app.ActivityThread.performDestroyActivity(ActivityThread.java:3693)
	... 10 more

```

ae3e, 2020-12-12:
So I finally moved to a more recent device.... with Android 10.
Now, mutliple shortcuts work perfectly fine...

closed 2020-12-12 by ae3e