victordiaz/PHONK #86: Phonk doesn't start when set a launch script

ae3e, 2020-11-15:
If i set a launch script in the settings, Phonk doesn't start any more and I get the following logs (I tried with a script having the `background_service` option set to true in `app.conf` and an another script with no `app.conf` file).

When Phonk doesn't start anymore, I can make it work again by edtiting the `settings.conf` file and remove the launch script.

```
java.lang.RuntimeException: Unable to start activity ComponentInfo{io.phonk.extended/io.phonk.MainActivity}: java.lang.ArrayIndexOutOfBoundsException: length=1; index=1
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2325)
	at android.app.ActivityThread.handleLaunchActivity(ActivityThread.java:2387)
	at android.app.ActivityThread.access$800(ActivityThread.java:151)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1303)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:135)
	at android.app.ActivityThread.main(ActivityThread.java:5254)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:902)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:697)
Caused by: java.lang.ArrayIndexOutOfBoundsException: length=1; index=1
	at io.phonk.runner.base.models.Project.<init>(Project.java:71)
	at io.phonk.MainActivity.onCreate(MainActivity.java:165)
	at android.app.Activity.performCreate(Activity.java:6033)
	at android.app.Instrumentation.callActivityOnCreate(Instrumentation.java:1106)
	at android.app.ActivityThread.performLaunchActivity(ActivityThread.java:2278)
	... 10 more
```

```
java.lang.RuntimeException: Unable to start service io.phonk.runner.AppRunnerLauncherService@3951d2e5 with null: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String android.content.Intent.getStringExtra(java.lang.String)' on a null object reference
	at android.app.ActivityThread.handleServiceArgs(ActivityThread.java:2911)
	at android.app.ActivityThread.access$2100(ActivityThread.java:151)
	at android.app.ActivityThread$H.handleMessage(ActivityThread.java:1401)
	at android.os.Handler.dispatchMessage(Handler.java:102)
	at android.os.Looper.loop(Looper.java:135)
	at android.app.ActivityThread.main(ActivityThread.java:5254)
	at java.lang.reflect.Method.invoke(Native Method)
	at java.lang.reflect.Method.invoke(Method.java:372)
	at com.android.internal.os.ZygoteInit$MethodAndArgsCaller.run(ZygoteInit.java:902)
	at com.android.internal.os.ZygoteInit.main(ZygoteInit.java:697)
Caused by: java.lang.NullPointerException: Attempt to invoke virtual method 'java.lang.String android.content.Intent.getStringExtra(java.lang.String)' on a null object reference
	at io.phonk.runner.AppRunnerLauncherService.onStartCommand(AppRunnerLauncherService.java:45)
	at android.app.ActivityThread.handleServiceArgs(ActivityThread.java:2894)
	... 9 more
```

Using PHONK 1.3.1_extended and Android 5.1.1

victordiaz, 2020-11-16:
Thanks for sharing this.
I have a few ideas of why it might not be working, I'll take this opportunity to add improve a bunch of stuff regarding the background services and so on.
I expect to release a new version by the end of this week and hopefully this will be fixed too :)

42loop, 2021-03-17:
while trying the launch script feature I still experience that phonk does not start any more - also i can't find documentation on the format of the string describing the script to be launched.
actually i was looking for a way to run phonk with a certain script on phone boot, so a student of mine could use it in an exhibition project. any ideas on this ?

42loop, 2021-03-17:
just found the "launch on device boot" option, but for me it does not work ...

victordiaz, 2021-03-17:
I had already some work on fixing the original issue but didn't find the time to complete the fix :sweat_smile: 
I'll have a look ASAP :)

42loop, 2021-03-23:
note: "launch on device boot" works on old android 5.1 phones, but not on android 10.
unfortunately i do not have others to test

victordiaz, 2021-03-23:
I've been working on the issue and things are shaping nicely, I just need a bit of time to do the final touches.

- @42loop Apparently Android changed some internals and now to launch an app on boot requires a permission that has to be enabled by the user manually. I think is called  "draw on top of apps" permission. 
In the new PHONK version the app will redirect you to the settings in Android so you can enable it.

- I added as well another option which is "launch a script on boot". So you can pick a script (with a nice dialog picker), which script to launch

victordiaz, 2021-04-02:
The freshly baked version [1.4.2](https://github.com/victordiaz/PHONK/releases/tag/1.4.2) has some improvements regarding this issue. It took some time! :sweat_smile: 
Feel free to try it out and let me know if it works in your devices :)

victordiaz, 2021-04-10:
I tried on android 8 and android 11 using PHONK 1.4.2 and works on both.

**Steps:**
I went to Settings > Script Launch on device Boot, selected a Project. Then rebooted the device. After unlocking the first time the selected script launches.

Maybe if somebody else could try it, we could find out what's going on.