victordiaz/PHONK #88: Permission to allow location acces with background scripts

ae3e, 2020-12-12:
Since Android 10, it looks like there's a specific permission for apps that run in background and need access to location : 

[ACCESS_BACKGROUND_LOCATION](https://developer.android.com/about/versions/10/privacy/changes) 

So unfortunately, one of my phonk's scripts is not working anymore after I migrated from android 5 to android 10. Would it be possible to add this new permission ?
Thanks

victordiaz, 2020-12-13:
Sure! I can add the permission.
I can create a 1.3.2 release tomorrow with the new permission so you check if it works.

I will ping you here once done :)
Thanks!

victordiaz, 2020-12-22:
Hi @ae3e 

I uploaded a new version 1.3.2 here https://github.com/victordiaz/PHONK/releases/tag/1.3.2
The permission is added, probably you will have to allow it from the Android Settings (there is a shortcut from PHONK settings now). Let me know if does work for you since I didnt try out myself :)

ae3e, 2020-12-24:
Thanks for this extra Christmas gift! 🎁 😄 
Just tested and it works well as before!

closed 2020-12-24 by ae3e