victordiaz/PHONK #89: Internal PNetwork.httpPost method not found

Slord6, 2020-12-20:
**Problem**
Calls to network.httpPost fail with console error: `The Method 'io.phonk.runner.apprunner.api.PNetwork.httpPost(string,object,function)' is not found`

**Device model**
Blackberry KeyOne

**Android version**
8.1.0

**Steps to reproduce the bug**
Run the `examples/Network/HTTP POST` example script

victordiaz, 2020-12-22:
Hi @Slord6,

Which version of PHONK are you using? 
In version 1.3.1 the http request functions were improved. Just make sure that your examples and/or scripts are updated in Settings -> Reinstall examples.

Can you check if that solves the problem?
I apologize if the examples updating process is not very streamlined yet

Slord6, 2020-12-22:
Hey, I've just done a fresh install from the play store to ensure I have the most up-to-date version but the 'About' section in the app's settings lists the version as `1.3.0_normal`. I'm still seeing the same issue, and the example scripts look the same - should I be seeing version 1.3.1 on the play store?

Also, as an aside, post re-install the app got into a bit of a weird state where it would crash trying to update the example scripts and couldn't create projects, but I solved that by clearing the app data (My scripts ended up reappearing too). I only mention this in case someone else has the same issue and searches the issue tracker.

victordiaz, 2020-12-22:
Oh! totally my bad. I forgot to upload 1.3.1 to the playstore 😅

Im uploading it as we speak (its in review now)

Btw, I just released 1.3.2 which you can find it here https://github.com/victordiaz/PHONK/releases/tag/1.3.2 
This new version will take some time to get it in the playstore. It seems that Google changed some policy stuff regarding a permission I just added here because https://github.com/victordiaz/PHONK/issues/88

Regarding the re-install. I had that too sometimes but I could not figure out the reason :/

Slord6, 2020-12-23:
Thanks @victordiaz, got the update installed and am now seeing the updated example script with the new method call.

However, if I create a new project and copy/paste the example http post code, I get an `okhttp3.internal.http2.ConnectionShutdownException` .

Full code, for clarity (but it's exactly the same as the example from what I can tell, and that runs fine):
```JavaScript
/*
 * PHONK Example: HTTP_POST
 *
 * With this code you are able to upload multipart content to a webserver
 * An example of a php file that will handle the upload is included
 *
 * The type of each data part can be text / json / file
 * If type === file then it will automatically load from the folder
 * and transfer
 *
 */

ui.addTitle(app.name)

network.httpRequest({
  method: 'POST',
  url: 'https://postman-echo.com/post',
  data: [
    { 'name': 'field_1', 'content': 'hello', 'type': 'text' },
    { 'name': 'field_2', 'content': 'world', 'type': 'text' },
    { 'name': 'userfile', 'content': 'patata.png', 'type': 'file', 'mediaType': 'image/png' }
  ]
}).onResponse(function (e) {
  console.log(e.status, e.response)
})

```

victordiaz, 2020-12-23:
Hi @Slord6 
Thanks for reporting the issue. I just replicated your steps and what I suspect is that when you copied the example to a different project. the file "patata.png" was not included and the error message is not quite meaningful.

Try to remove the following line and let me know if it works
` { 'name': 'userfile', 'content': 'patata.png', 'type': 'file', 'mediaType': 'image/png' }`

I'll try to catch the exception and give better feedback on the console

Slord6, 2020-12-23:
Ah, yes - of course! Thanks!

closed 2020-12-23 by Slord6

victordiaz, 2020-12-23:
I'm crrating a new issue with the feedback you gave me :)