victordiaz/PHONK #94: Add allowing generic HTTP POST body

Slord6, 2020-12-23:
As per discussion #92 , it would be useful to be able to have control over the full body of HTTP POST requests.

For example, I have a simple [HTTP wrapper](https://github.com/Slord6/generic-data-store) around a JSON-backed database, which expects POST requests to contain only the data to be added to the database. At the moment it is therefore not possible to POST data with Phonk to that end-point.

Example code + expected data at discussion #92

victordiaz, 2020-12-24:
@Slord6 I uploaded a new version here for you to test https://github.com/victordiaz/PHONK/releases/tag/1.3.3pre
Let me know how it goes :)

Slord6, 2020-12-24:
That's working great. Thanks!

SeriousM, 2021-01-29:
Is there any chance to get a new release with this enhancements included?

Slord6, 2021-01-29:
I think it was already released. In 1.3.1 looking at the release notes. My last use of it looks like:

```
const sendData = (data) => {
  data.client = "Phone";
  data.satCount = satCount;
  console.log("Sending: " + JSON.stringify(data));
  network.httpRequest({
   method: 'POST',
  url: url,
  body: {
    type: 'application/json',
    data: JSON.stringify(data)
  }}).onResponse(function (e) {
    console.log(e.status, e.response);
  })
}
```

victordiaz, 2021-01-29:
it should work on 1.40 too unless I screwed up at some point upgrading stuff :)

@Slord6 btw thx! you just made me realize with your code that at some point Mozilla Rhino interpreter got fat arrows and const!!!!
I didnt know at all!.
On top of that I just realized that toggling a flag we can finally have `let` and destructuring support :)

I'll release a new version during the weekend, thx for sharing that code!

Slord6, 2021-01-29:
Oh nice, it hadn't even occurred to me they weren't supported so glad it's helpful!

SeriousM, 2021-01-30:
@victordiaz thanks for looking into it.

In v1.3.3pre I had the option to use 
```
  network.httpRequest({
    method: 'POST',
    url: 'https://bzzzzzzz.com/bzzzzz',
    body: {
      type: 'application/json',
      data: JSON.stringify([input.text()])
    }
  }).onResponse(function (e) {
```

but with v1.4.0 I get the error `java.lang.IllegalArgumentException: method POST must have a request body.` as with previous versions.

I hope you can include the specific code in the upcoming version!
Thank you very much for this wonderful framework :)

victordiaz, 2021-01-30:
@SeriousM you are right, it does not work currently. It seems I didnt merge properly :sweat_smile: 
I'll try to upload a new version today or tomorrow fixing the issue.