victordiaz/PHONK #113: No example for sms send/receive

t-oster, 2022-05-31:
Hi,
Nice app. Unfortunately I cannot find any information regarding the SMS API. I want to build a simple SMS to MQTT gateway (bidirectional), which should be very easy with Phonk. Can you point me to the docs or the sources with the SMS functionality?

polhomarkho, 2022-06-01:
Hi!  

The SMS functions are defined here `phonk_apprunner/src/main/java/io/phonk/runner/apprunner/api/PDevice.java`: `device.onSmsReceived()` and `device.smsSend()`.
Here is a small example that sends and receives a SMS. The phone number is asked when running:

```javascript
/*	
 *  Description Send and receive a SMS
 *  by Paul-Emile Sublet <polhomarkho@gmail.com>
 */

ui.addTitle(app.name);

var phoneNumber;

const receivedSmsDetails = ui.addText(0, 0, 1, 0.5);
ui.popup()
  .title('Enter the phone number to send an SMS to')
  .ok('ok')
  .input('phone number')
  .onAction(function(phoneNumberInput) {
    phoneNumber = phoneNumberInput.answer;
  })
  .show();

// these methods are defined in phonk_apprunner/src/main/java/io/phonk/runner/apprunner/api/PDevice.java
device.onSmsReceived(function(sms) {
  receivedSmsDetails.text('from: ' + sms.from + "\n" + 'message: ' + sms.message);
});

ui.addButton('send sms', 0, 0.5, 1, 0.5)
  .onClick(function() {
    device.smsSend(phoneNumber, 'hello world!');
  });
```

polhomarkho, 2022-06-29:
Here is the PR adding this example: https://github.com/victordiaz/PHONK/pull/119