victordiaz/PHONK #132: TypeScript declaration

notEvil, 2023-01-10:
Hi,

I've been using the web editor but find the documentation to be outdated and the lack of autocompletion makes it unnecessarily difficult to get started.

So I looked for an alternative and found TS declaration files. The following script reads the relevant Java files in this repository and creates declarations which in turn provide type inference and autocompletion.

https://gitlab.com/-/snippets/2482545

In Neovim with Ycm
[image]
[image]

This is merely a proof-of-concept, expect wrong/generic types, missing method arguments, class boundary violations (nested classes), ...
I'm not familiar with Java, so I won't go any further with this, but you should have no trouble adjusting your documentation export to produce something along these lines.

notEvil, 2023-01-10:
Nevermind, saw https://github.com/c2nes/javalang and couldn't resist: https://github.com/notEvil/PHONK/blob/ts_declarations/get_ts_declarations.py
Now I would consider this actually viable. It still assumes a lot and is far from perfect, but has a solid foundation.

For people not familiar with Python, the easiest way to run this is:
`python -m pip install pipenv; python -m pipenv run pip install javalang; python -m pipenv run python ./get_ts_declarations.py .`

victordiaz, 2023-01-12:
Hi @notEvil this sounds really amazing! This for sure will help to update the old docs,
I will have to check how to integrate it in the ace editor that PHONK uses.

I'm a bit busy atm but I will try this and answer your other questions by the end of next week.
Thanks!

notEvil, 2023-01-12:
Great, no hurry! I now use sshfs and Neovim as shown above, and it works quite well.

Fyi I've changed the script a bit so it would export the documentation comments :)