victordiaz/PHONK #137: Widget TextList: text size doesn't apply

notEvil, 2023-02-24:
Hi,

```javascript
var text_list = ui.addTextList(0, 0, 1, 1)
text_list.textSize(10)
text_list.autoScroll(true)

var count = 0

util.loop(1000, () => {
  text_list.add((count++).toString())
}).start()
```

gives me

[image]

All lines except one are using the default text size. It is semi random which line is using the correct text size. I have no clue why; an implementation based on List works as expected, see https://gitlab.com/notEvil/cyclometer/-/blob/master/PHONK/app/lib/mui.js.

victordiaz, 2023-02-24:
oh! I will have a look.
I allocated time next week to do some PHONK work.

By the way, feel free to join the Discord channel in case you wanna have a chat there!

notEvil, 2023-03-12:
fixed in https://github.com/victordiaz/PHONK/pull/141

closed 2023-03-12 by notEvil